littlekernel/lk #54: QEMU timer not firing

cpu-chromium, 2015-09-24:
For build-qemu-virt-a15  which is executed in qemu-system-arm -machine virt -cpu cortex-a15 might not fire in some versions. For example for QEMU 2.4.50 the following patch fixes it:

index d7cb9f4..52d0b9d 100644
--- a/platform/qemu-virt/include/platform/qemu-virt.h
+++ b/platform/qemu-virt/include/platform/qemu-virt.h
@@ -88,12 +88,8 @@ static const int a15irqmap[] = {
 
 /* interrupts */
 #define ARM_GENERIC_TIMER_VIRTUAL_INT 27
-#if ARCH_ARM
-/* for cortex-a15, the irq seems to be coming in via the secure vector */
-#define ARM_GENERIC_TIMER_PHYSICAL_INT 29
-#else
+/* Note that in some QEMUS the physical timer is wired to INT 29 */
 #define ARM_GENERIC_TIMER_PHYSICAL_INT 30
-#endif
 #define UART0_INT   (32 + 1)
 #define VIRTIO0_INT (32 + 16)

The current QEMU behavior, which is the timer mapped to 30 seems to start end of may, or possibly related to the default mode of virt which used to be trustzone, which is recent:

https://github.com/qemu/qemu/commit/2d710006a0da4a9b7ddf5c02d072e178906d0ef6

travisg, 2016-01-18:
There are a few ways to fix this. One is to ignore it and assume all versions of qemu are new. Another is I think to read the irq number out of the device tree passed it by qemu, though would also need to make sure those are valid in old versions of qemu.

sandeep-mvd, 2016-03-12:
Hi @travisg 
Can you assign this issue to me? I will look into it and fix it.
This is going to be my first commit and so I might take a little bit more time and also might need some help with the process.

Thank you.

travisg, 2016-03-15:
Sure give it a go. I think the irq can be read from the FDT that qemu passes to the OS.

I can't assign you to this problem for some reason though.

sandeep-mvd, 2016-03-17:
Thank you.. I am wrking on it.

sandeep-mvd, 2016-03-21:
Sorry if this is a basic question. But I cannot see any dtsi files located in the source tree(generally i though they would b e present arm/arm64/boot/dts/). Where are these located?

Also kindly point me to any reference documents, if available, that would help the first timers to lk/qemu like me.
Thank you.

travisg, 2016-03-23:
The device tree is passed to the kernel from qemu and generated dynamically at run time by qemu. If i remember, there is an optional switch on qemu to tell it to dump the device tree to a file to be inspected.