littlekernel/lk #66: [x86_64][MMU] should not check canonical form addresses for physical address

bzhu5, 2015-11-02:
According to this site below, canonical address is meaningful only for VIRTUAL address. 
https://en.wikipedia.org/wiki/X86-64#Canonical_form_addresses

However, in this function: lk/arch/x86-64/mmu.c:x86_mmu_add_mapping(), as below.

if((!x86_mmu_check_map_addr(vaddr)) || (!x86_mmu_check_map_addr(paddr)) )
		return ERR_INVALID_ARGS;

It is not only checking virtual address but also checking physical address. 

Besides, there are many other places to check both virtual and physical by calling x86_mmu_check_map_addr().

In my opinion, for virtual address , it is ok to check canonical form, but for physical address , just check it should be less than max supported address reported by CPUID capability.

travisg, 2015-11-03:
I agree. If you'd like to fix it I'd gladly take the change.

bzhu5, 2015-11-04:
Ok, i can do that recently.

travisg, 2015-11-06:
merged upstream

closed 2015-11-06 by travisg