littlekernel/lk #71: x86-64 fpu context save seems broken

travisg, 2015-11-09:
The x86-64 fpu context saving seems broken. float_tests, which runs a bunch of separate threads doing calculations that should result in similar answers gets different results every time:

] float_tests
floating point test:
creating 8 floating point threads
float thread 0 returns 1, val 821434452.497358
float thread 1 returns 1, val 81419905.943663
float thread 2 returns 1, val 29789586.236013
float thread 3 returns 1, val 30216835.564976
float thread 4 returns 1, val 227027182.938372
float thread 5 returns 1, val 886723275.849505
float thread 6 returns 1, val 303244330.207326
float thread 7 returns 1, val 222952286.814904
] float_tests
floating point test:
creating 8 floating point threads
float thread 0 returns 1, val 240072320.917543
float thread 1 returns 1, val 1092221311.715492
float thread 2 returns 1, val 660033076.162435
float thread 3 returns 1, val 1102409849.573568
float thread 4 returns 1, val 1008718492.368174
float thread 5 returns 1, val 299878691.231605
float thread 6 returns 1, val 15341624.298121
float thread 7 returns 1, val 601862821.592540

on x86-32 and arm the results are very close and consistent:
] float_tests
floating point test:
creating 8 floating point threads
float thread 0 returns 1, val 2811768340480.000000
float thread 1 returns 1, val 2811768340480.000000
float thread 2 returns 1, val 2811768602624.000000
float thread 3 returns 1, val 2811768602624.000000
float thread 4 returns 1, val 2811768602624.000000
float thread 5 returns 1, val 2811768602624.000000
float thread 6 returns 1, val 2811768602624.000000
float thread 7 returns 1, val 2811768602624.000000

This seems to have nothing to do with the recent x86-64 integer context switch changes, it was broken before those changes went through.

bzhu5, 2015-11-10:
I saw the same issue on Qemu.

and the "float_tests" runs 8 threads to do calculations, however, if I changed the thread number as ONE only. Everytime I run it for 64bit and even get the **different** result.  This doesn't make sense!

Does Qemu have issues for x64 FPU support?  Not sure if this exists on PC.

bzhu5, 2015-11-10:
for the code lk\app\tests\float.c: float_thread() 
I found x86 and x86_64 generate different asm code. 
* x86_64: it uses xmm0~xmm15
* x86: it only use st0~st7

it is also weird that _printf() is using xmm0~7 (saving them) for x86_64.

Besides, there are other findings:
* if we add a line like **printf("\n")** in this function **float_thread()**, then this issue disappears
* even don't do any context switch, I mean change the code to make LK only has **one** thread (idle), the issue still happens whenever the interrupt is enabled (sti). (interrupt enable means timer interrupt can happen every a few mseconds)
* if a the float test routine only uses xmm0~xmm7, then issue disappears.

bzhu5, 2015-11-10:
I suspect this is a x64 compiler bug when it generating float_thread() code with **O3**.

bzhu5, 2016-03-03:
looks like this issue has been fixed in the latest code.

closed 2016-03-15 by travisg