littlekernel/lk #121: dlmalloc.c:1752:44: error: implicit declaration of function ‘mremap’

Xoooo, 2016-03-13:
ubutu64@ubutu64:~/lk$ make pc-x86-64-test
make[1]: Entering directory '/home/ubutu64/lk'
make[2]: Entering directory '/home/ubutu64/lk'
PROJECT = pc-x86-64-test
PLATFORM = pc
TARGET = pc-x86
arch/x86/rules.mk:73: ARCH_x86_TOOLCHAIN_PREFIX = 
arch/x86/rules.mk:74: ARCH_x86_64_TOOLCHAIN_PREFIX = 
arch/x86/rules.mk:75: TOOLCHAIN_PREFIX = 
including app app/shell app/stringtests app/tests dev kernel lib/aes lib/aes/test lib/cbuf lib/cksum lib/debugcommands lib/libm lib/version platform target
including kernel/vm lib/console lib/debug lib/heap lib/libc
including lib/heap/dlmalloc lib/io
LIBGCC = /usr/lib/gcc/x86_64-linux-gnu/5/libgcc.a
GLOBAL_COMPILEFLAGS = -g -finline -include ./build-pc-x86-64-test/config.h -W -Wall -Wno-multichar -Wno-unused-parameter -Wno-unused-function -Wno-unused-label -Werror=return-type -fasynchronous-unwind-tables -gdwarf-2 -fno-pic -fno-stack-protector -mcmodel=kernel -mno-red-zone
GLOBAL_OPTFLAGS = -O2
generating build-pc-x86-64-test/target/pc-x86/module_config.h
generating build-pc-x86-64-test/config.h
generating build-pc-x86-64-test/platform/pc/module_config.h
generating build-pc-x86-64-test/arch/x86/module_config.h
generating build-pc-x86-64-test/top/module_config.h
generating build-pc-x86-64-test/app/module_config.h
generating build-pc-x86-64-test/app/shell/module_config.h
generating build-pc-x86-64-test/app/stringtests/module_config.h
generating build-pc-x86-64-test/app/tests/module_config.h
generating build-pc-x86-64-test/dev/module_config.h
generating build-pc-x86-64-test/kernel/module_config.h
generating build-pc-x86-64-test/external/lib/aes/module_config.h
generating build-pc-x86-64-test/external/lib/aes/test/module_config.h
generating build-pc-x86-64-test/lib/cbuf/module_config.h
generating build-pc-x86-64-test/external/lib/cksum/module_config.h
generating build-pc-x86-64-test/lib/debugcommands/module_config.h
generating build-pc-x86-64-test/external/lib/libm/module_config.h
generating build-pc-x86-64-test/srcfiles.txt
generating build-pc-x86-64-test/include_paths.txt
generating build-pc-x86-64-test/buildid.h
build-pc-x86-64-test/buildid.h.tmp build-pc-x86-64-test/buildid.h 不同：第 108 字节，第 3 行
generating build-pc-x86-64-test/lib/version/module_config.h
compiling lib/version/version.c
linking build-pc-x86-64-test/lib/version.mod.o
generating build-pc-x86-64-test/platform/module_config.h
generating build-pc-x86-64-test/target/module_config.h
generating build-pc-x86-64-test/kernel/vm/module_config.h
generating build-pc-x86-64-test/lib/console/module_config.h
generating build-pc-x86-64-test/lib/debug/module_config.h
generating build-pc-x86-64-test/lib/heap/module_config.h
generating build-pc-x86-64-test/lib/libc/module_config.h
generating build-pc-x86-64-test/external/lib/heap/dlmalloc/module_config.h
compiling external/lib/heap/dlmalloc/dlmalloc.c
external/lib/heap/dlmalloc/dlmalloc.c: In function ‘mmap_resize’:
external/lib/heap/dlmalloc/dlmalloc.c:1752:44: error: implicit declaration of function ‘mremap’ [-Werror=implicit-function-declaration]
 #define MREMAP_DEFAULT(addr, osz, nsz, mv) mremap((addr), (osz), (nsz), (mv))
                                            ^
external/lib/heap/dlmalloc/dlmalloc.c:1808:49: note: in expansion of macro ‘MREMAP_DEFAULT’
         #define CALL_MREMAP(addr, osz, nsz, mv) MREMAP_DEFAULT((addr), (osz), (
                                                 ^
external/lib/heap/dlmalloc/dlmalloc.c:3924:23: note: in expansion of macro ‘CALL_MREMAP’
     char* cp = (char*)CALL_MREMAP((char*)oldp - offset,
                       ^
external/lib/heap/dlmalloc/dlmalloc.c:3924:16: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     char* cp = (char*)CALL_MREMAP((char*)oldp - offset,
                ^
external/lib/heap/dlmalloc/dlmalloc.c: In function ‘sys_trim’:
external/lib/heap/dlmalloc/dlmalloc.c:4369:62: warning: comparison between pointer and integer
             if ((CALL_MREMAP(sp->base, sp->size, newsize, 0) != MFAIL) ||
                                                              ^
cc1: some warnings being treated as errors
make/compile.mk:41: recipe for target 'build-pc-x86-64-test/external/lib/heap/dlmalloc/dlmalloc.o' failed
make[2]: *** [build-pc-x86-64-test/external/lib/heap/dlmalloc/dlmalloc.o] Error 1
make[2]: Leaving directory '/home/ubutu64/lk'
engine.mk:28: recipe for target 'make-make' failed
make[1]: *** [make-make] Error 2
make[1]: Leaving directory '/home/ubutu64/lk'
makefile:34: recipe for target 'pc-x86-64-test' failed
make: *** [pc-x86-64-test] Error 2

travisg, 2016-03-13:
You're using the linux toolchain, which is not supported. I think it's confusing the dlmalloc code, which thinks it is building for linux.

bzhu5, 2016-03-14:
yes, looks like toolchain issue, i didn't have this build issue.

nvll, 2016-03-24:
Let us know if this or a similar issue persists with the correct toolchain.

closed 2016-03-24 by nvll

egrimley-arm, 2020-05-01:
For what it's worth, I was able to build with `aarch64-linux-gnu-gcc` by inserting `#undef linux` at the start of `external/lib/heap/dlmalloc/dlmalloc.c`. Since `aarch64-linux-gnu-gcc` is part of Debian and Ubuntu, using the wrong toolchain might be easier for some people. See also #195.

moe123, 2020-05-01:
that's a classic mixing up OSTYPE platforms with host/target specifics; you should never rely on generic-vendor targets.

travisg, 2020-05-02:
Probably fairly harmless to undef the linux thing, though clearly it's a hack. Too bad it's actually set by the compiler. Wonder if it's possible to -Ulinux it on the command line, if the build system detects a linux toolchain...