littlekernel/lk #148: patch for makecontext, swapcontext, getcontext and setcontext required for lk/lib/libc

rkbug, 2016-07-26:
I am porting an x86 SW code into ARM based system. The SW has implemented threading functions using (makecontext, swapcontext, getcontext and setcontext) routines. These routines are missing in the ~/lk/lib/libc code. Can anyone help me with the patch for these routines?

closed 2018-01-10 by rkbug

travisg, 2018-01-10:
Ah guess I lost track of this one. Not entirely sure what those do. Are they like setjmp/longjmp?

M1cha, 2018-01-10:
http://man7.org/linux/man-pages/man3/makecontext.3.html
https://linux.die.net/man/2/setcontext

Basically it's setjmp/longjmp with an API optimized for threading because you can easily return to the old context, change the stack pointer and the entrypoint, etc.

On Unix systems it actually does more than setjmp/longjmp because it saves more data like the signal mask so you can set that from within a signal handler to allow external/timed scheduling.