littlekernel/lk #172: Incorrect cache line size in arm64 architecture.

mazhenke, 2016-10-09:
In lk/arch/arm64/include/arch/defines.h file:

#define CACHE_LINE 32

But actually the cortex-a53, cortex-a57 and cortex-a72 cache line size is **64** bytes.

travisg, 2016-10-09:
You're right. Throw together a patch and I'll commit it.

kyle-github, 2017-06-15:
This is a fairly old bug.   Please let me know if I should create another instead of adding to it.

Some ARM64 hardware does not have a single cache line size.  bigLITTLE configurations can have one cache line size for one set of cores and a different one for the other set.  See below:

http://www.mono-project.com/news/2016/09/12/arm64-icache/

The above link shows an example of this on Exynos systems.  For most of the uses I find in the code (alignment) it seems like going with the larger size would work, but for cache line clearing you may run into the same bug the Mono guys did.   

Oh, and Exynos cache line sizes are 64 for the little cores and 128 for the big ones.  At least that version of Exynos.

travisg, 2017-06-23:
Will update. I've already fixed this on this magenta side of things (which forks from this project). At some point I'll probably switch it over to some sort of detection mechanism, but since LK primarily runs on thigns that is is compiled to, there hasn't been a big desire.