littlekernel/lk #184: Snapdragon 820 MSM8996

umerov1999, 2017-04-17:
Can I flash lk bootloader for Snapdragon 820 MSM8996 on ZTE Axon 7? This Smartphone using verificate bootloader(Проверяет загрузчик по x509 ключу)

M1cha, 2017-04-22:
This is not the code, Qualcomm uses for LK, they use: https://www.codeaurora.org/cgit/quic/la/kernel/lk
Also, you usually can't replace the existing LK, because it's signed. You'd have to chainload it from fastboot or from the boot/recovery partition.

dhimanchakraborty, 2017-08-26:
Is there any document available to build qualcomm little kernel and chainload it from fastboot or from the boot/recovery partition?

travisg, 2017-08-26:
No, this project is unrelated to the qualcomm bootloader. See previous comments.

closed 2017-08-26 by travisg