littlekernel/lk #187: What is the meaning of dpc system?

takboo, 2017-05-16:
Recently i was study little kernel project, i'm curious the full name of dpc and where this system to use, thanks.

moe123, 2017-05-16:
@hello takboo, it looks like you are using a translator tool or something ... likewise, may you refactor and give a bit of context ; if you are talking about kernel/dpc:
```
-> "deferred procedure call" if I am not mistaken
	-> you have thread holding a queue of events
		-> then when thread runs
			-> for each events
				-> dispatch
```

M1cha, 2017-05-16:
Afaik @moe123 is right. dpc is used to make actions run later. In past it was used to free the current thread's stack when it exits but thread.c uses its own implementation now and dpc is unused.

takboo, 2017-05-17:
@moe123 @M1cha thanks for your help. 
English is not my first language, if I use wrong words, please forgive me.
kernel/dpc is the dpc i want to know.
 I found dpc_init in kmain(), i can understand the code, but i don't know the purpose for "dpc", so i 
ask this question.
again, thanks for your help.

travisg, 2017-05-17:
DPC stands for 'deferred procedure call'. It's currently unused in LK, but
it basically is a simple little mechanism to force a function to be called
on the context of a worker thread.

M1cha, 2017-05-17:
@travisg is there unix-like terminology/alternative for this too? Googling dpc only gives you results for Microsofts DPC.

zeusk, 2017-05-17:
@M1cha, tasklets and bottom half are used in linux in a pretty similar manner.

moe123, 2017-05-17:
@M1cha, @zeusk, yes there is Tasklets ; but this is more complex than simply delaying a bit some calls, indeed Tasklets are `interrupts aware`, priorities. DPC is meant to add a bit of latency, crossing your fingers and letting it go, a really simple and lightweight system,  if always called from the same context, DPC handles the `job` well. You can find similar approaches in other kernels and kernel APIs such as network layers.