littlekernel/lk #195: compile error issue

prime-zeng, 2017-07-08:
external/lib/heap/dlmalloc/dlmalloc.c: In function ‘mmap_resize’:
external/lib/heap/dlmalloc/dlmalloc.c:1752:44: error: implicit declaration of function ‘mremap’ [-Werror=implicit-function-declaration]
 #define MREMAP_DEFAULT(addr, osz, nsz, mv) mremap((addr), (osz), (nsz), (mv))

ARCH_arm_TOOLCHAIN_PREFIX=arm-linux-gnueabihf- ./scripts/do-qemuarm

travisg, 2017-07-08:
I've seen that one before, it's something to do with the linux toolchain setting some #defines that force the dlmalloc code into thinking it's compiling for linux. I can't debug it on this end, but the lines it's failing on is probably not supposed to be compiled.

Officially, LK is only supported building from a generic -elf or -eabi toolchain, specifically because of this sort of stuff. Using a linux toolchain tends to cause a lot of trouble, this is a classic example of it.

prime-zeng, 2017-07-08:
I have just tried the eabi toolchain, and it works, thank you. By the  way,  i am new to LK, and i am want to study it , any suggestion about where to start from, is the LK comatible with the GP TEE?

egrimley-arm, 2020-05-01:
For what it's worth, I was able to build with `aarch64-linux-gnu-gcc` by inserting `#undef linux` at the start of `external/lib/heap/dlmalloc/dlmalloc.c`. Since `aarch64-linux-gnu-gcc` is part of Debian and Ubuntu, using the wrong toolchain might be easier for some people. See also #121.

echoptic, 2023-04-29:
I also had this issue while compiling for riscv64 using the compiler thats recommended in the getting started guide and
adding `#undef linux` fixed the problem. Are you considering adding that fix?

travisg, 2023-05-02:
In general it's kinda a mess for folks to use the linux toolchain with LK,
but if it's a straightforward easy fix it's probably worth trying. Of
course it then boils down to precisely which version of which toolchain,
etc and it's a real pain to maintain. I can maybe give it a try though.