littlekernel/lk #198: test run for "make lm3s6965evb-test"

wenhuizhang, 2017-08-08:
Hi, I was trying to give it a test run for cortex-m3, however got error trace as following, any idea of why this happens please?

```
vagrant@vagrant-ubuntu-trusty-32:~/lk/build-lm3s6965evb-test$ qemu-system-arm -machine versatileab -cpu cortex-m3 ./lk.elf
pulseaudio: pa_context_connect() failed
pulseaudio: Reason: Connection refused
pulseaudio: Failed to initialize PA contextaudio: Could not init `pa' audio driver
ALSA lib confmisc.c:768:(parse_card) cannot find card '0'
ALSA lib conf.c:4248:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4248:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4248:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4727:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default
alsa: Could not initialize DAC
alsa: Failed to open `default':
alsa: Reason: No such file or directory
ALSA lib confmisc.c:768:(parse_card) cannot find card '0'
ALSA lib conf.c:4248:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4248:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4248:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4727:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default
alsa: Could not initialize DAC
alsa: Failed to open `default':
alsa: Reason: No such file or directory
audio: Failed to create voice `lm4549.out'
Could not initialize SDL(No available video device) - exiting
```

wenhuizhang, 2017-08-09:
I have updated qemu-system-arm with 2.9 version. And using `qemu-system-arm -machine versatileab -cpu cortex-m4 ./lk.bin` I got:

```
vagrant@vagrant-ubuntu-trusty-32:~/lk/build-lm3s6965evb-test$ qemu-system-arm -machine versatileab -cpu cortex-m4 ./lk.elf
WARNING: Image format was not specified for './lk.elf' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
audio: Could not init `oss' audio driver
gtk initialization failed
```
Any suggestions for debugging traces please?

nvll, 2017-08-09:
These are issues with your Linux system unrelated to LK. Qemu is trying to
use 'oss' as the audio background and is unable to. You'll need to specify
the correct one to use, or figure out why 'oss' isn't working.

wenhuizhang, 2017-08-09:
What happens if I am using a container with no audio and no graphical
interfaces... I think I am not using gtk . Any suggestions please?

wenhuizhang, 2017-08-09:
Is it possible to eliminate audio  and graphics interfaces modules on little kernel while compiling please? I think I only need file system and network interfaces/the tcp/ip stack.

wenhuizhang, 2017-08-09:
Hi, I think it is just a mapping issue, however I am not sure m3 repo stands for which board and which cpu please?

closed 2017-08-09 by wenhuizhang