littlekernel/lk #199: Halt == spinning forever

wenhuizhang, 2017-08-09:
When close using "poweroff", it does not close the qemu socket

travisg, 2017-08-10:
Which qemu target were you running? It's possible that not all of the poweroff paths are implemented for every target.

antoneliasson, 2017-08-10:
Reproducible on ARM (per the instructions in the README). Also happens for the reboot command

travisg, 2017-08-10:
Ah yes that's just because the platform_halt() path isn't implemented on all of the platforms, and the default one just drops to a spin like that. It doesn't seem to be implemented in the qemu emulated ones, so that's the behavior that's expected. I'd be easy enough to add it for the qemu target.

wenhuizhang, 2017-08-13:
Qemu lm3s6965evb-test and also nearly all qemu arm versions

travisg, 2017-08-13:
Correct. It has nothing at all to do with qemu, it's because the individual platform/target code in LK doesn't necessarily have the final mechanism to shut down the emulated system, so it has a fallback path that just spins the cpu.