littlekernel/lk #200: arp list crashes LK

antoneliasson, 2017-08-10:
Commit  63fa995, Ubuntu 16.04 host.

```
[5.1.1]antone@lnxantone1:~/git/lk> scripts/do-qemuarm
DO_BLOCK = 0
DO_NET = 0
make[1]: Entering directory '/home/antone/git/lk'
make[2]: Entering directory '/home/antone/git/lk'
PROJECT = qemu-virt-a15-test
PLATFORM = qemu-virt
TARGET = qemu-virt

TOOLCHAIN_PREFIX = arm-none-eabi-
LIBGCC = /usr/lib/gcc/arm-none-eabi/4.9.3/armv7-ar/thumb/softfp/libgcc.a
GLOBAL_COMPILEFLAGS = -g -finline -include ./build-qemu-virt-a15-test/config.h -W -Wall -Wno-multichar -Wno-unused-parameter -Wno-unused-function -Wno-unused-label -Werror=return-type -fno-common -mthumb-interwork  -mcpu=cortex-a15 -mfpu=vfpv3 -mfloat-abi=softfp -mthumb -D__thumb__
including app app/inetsrv app/shell app/stringtests app/tests dev dev/interrupt/arm_gic dev/timer/arm_generic dev/virtio/block dev/virtio/gpu dev/virtio/net kernel lib/aes lib/aes/test lib/cbuf lib/cksum lib/debugcommands lib/fdt lib/fs lib/fs/ext2 lib/fs/fat32 lib/fs/memfs lib/fs/spifs lib/fs/spifs/test lib/libm lib/minip lib/version platform target
including dev/virtio kernel/vm lib/bcache lib/bio lib/console lib/debug lib/fixed_point lib/gfx lib/heap lib/iovec lib/libc lib/pool lib/tftp
including lib/heap/dlmalloc lib/io
LIBGCC = /usr/lib/gcc/arm-none-eabi/4.9.3/armv7-ar/thumb/softfp/libgcc.a
GLOBAL_COMPILEFLAGS = -g -finline -include ./build-qemu-virt-a15-test/config.h -W -Wall -Wno-multichar -Wno-unused-parameter -Wno-unused-function -Wno-unused-label -Werror=return-type -fno-common -mthumb-interwork
GLOBAL_OPTFLAGS = -O2
generating build-qemu-virt-a15-test/platform/qemu-virt/module_config.h
generating build-qemu-virt-a15-test/config.h
generating build-qemu-virt-a15-test/top/module_config.h
generating build-qemu-virt-a15-test/arch/arm/module_config.h
generating build-qemu-virt-a15-test/app/inetsrv/module_config.h
generating build-qemu-virt-a15-test/app/module_config.h
generating build-qemu-virt-a15-test/app/shell/module_config.h
generating build-qemu-virt-a15-test/app/stringtests/module_config.h
generating build-qemu-virt-a15-test/app/tests/module_config.h
generating build-qemu-virt-a15-test/dev/module_config.h
generating build-qemu-virt-a15-test/dev/interrupt/arm_gic/module_config.h
generating build-qemu-virt-a15-test/dev/timer/arm_generic/module_config.h
generating build-qemu-virt-a15-test/dev/virtio/block/module_config.h
generating build-qemu-virt-a15-test/dev/virtio/gpu/module_config.h
generating build-qemu-virt-a15-test/dev/virtio/net/module_config.h
generating build-qemu-virt-a15-test/kernel/module_config.h
generating build-qemu-virt-a15-test/external/lib/aes/module_config.h
generating build-qemu-virt-a15-test/external/lib/aes/test/module_config.h
generating build-qemu-virt-a15-test/lib/cbuf/module_config.h
generating build-qemu-virt-a15-test/external/lib/cksum/module_config.h
generating build-qemu-virt-a15-test/lib/debugcommands/module_config.h
generating build-qemu-virt-a15-test/external/lib/fdt/module_config.h
generating build-qemu-virt-a15-test/lib/fs/module_config.h
generating build-qemu-virt-a15-test/lib/fs/ext2/module_config.h
generating build-qemu-virt-a15-test/lib/fs/fat32/module_config.h
generating build-qemu-virt-a15-test/lib/fs/memfs/module_config.h
generating build-qemu-virt-a15-test/lib/fs/spifs/module_config.h
generating build-qemu-virt-a15-test/lib/fs/spifs/test/module_config.h
generating build-qemu-virt-a15-test/external/lib/libm/module_config.h
generating build-qemu-virt-a15-test/lib/minip/module_config.h
generating build-qemu-virt-a15-test/buildid.h
build-qemu-virt-a15-test/buildid.h.tmp build-qemu-virt-a15-test/buildid.h differ: byte 115, line 3
generating build-qemu-virt-a15-test/lib/version/module_config.h
generating build-qemu-virt-a15-test/platform/module_config.h
generating build-qemu-virt-a15-test/target/module_config.h
generating build-qemu-virt-a15-test/dev/virtio/module_config.h
generating build-qemu-virt-a15-test/kernel/vm/module_config.h
generating build-qemu-virt-a15-test/lib/bcache/module_config.h
generating build-qemu-virt-a15-test/lib/bio/module_config.h
generating build-qemu-virt-a15-test/lib/console/module_config.h
generating build-qemu-virt-a15-test/lib/debug/module_config.h
generating build-qemu-virt-a15-test/lib/fixed_point/module_config.h
generating build-qemu-virt-a15-test/lib/gfx/module_config.h
generating build-qemu-virt-a15-test/lib/heap/module_config.h
generating build-qemu-virt-a15-test/lib/iovec/module_config.h
generating build-qemu-virt-a15-test/lib/libc/module_config.h
generating build-qemu-virt-a15-test/lib/pool/module_config.h
generating build-qemu-virt-a15-test/lib/tftp/module_config.h
generating build-qemu-virt-a15-test/external/lib/heap/dlmalloc/module_config.h
[... 108 lines cut ...]
r0  0x00000000 r1  0x800453c0 r2  0x00000000 r3  0x00000000
r4  0x00000000 r5  0x801e07e0 r6  0x800594bc r7  0x800593e8
r8  0x801e0760 r9  0x801e0758 r10 0x00000008 r11 0x801e0928
r12 0x7a0b5ed8 usp 0x00000000 ulr 0x00000000 pc  0x8002b7be
spsr 0x40000173
 usr r13 0x00000000 r14 0x00000000
 fiq r13 0x00000000 r14 0x00000000
 irq r13 0x00000000 r14 0x800122d0
asvc r13 0x801e05c8 r14 0x80010a8b
*svc r13 0x801e0698 r14 0x8002c0c7
 und r13 0x00000000 r14 0x00000000
 sys r13 0x00000000 r14 0x00000000
bottom of stack at 0x801e0698:
0x801e0698: 00000000 80048450 00000061 00000008 |....P...a.......|
0x801e06a8: 800594bc 801dc640 00000000 801e07fc |....@...........|
0x801e06b8: 00000001 00000000 801e07e0 800594bc |................|
0x801e06c8: 00000000 8002c0c7 8002c051 00000000 |........Q.......|
0x801e06d8: 800594bc 80033a55 00009801 80039533 |....U:......3...|
0x801e06e8: 99999999 801e07e0 80033c07 00000001 |.........<......|
0x801e06f8: 80033375 00000000 00000000 800594bc |u3..............|
0x801e0708: 00000000 801e0758 00000000 800594bc |....X...........|
CRASH: starting debug shell... (reason = 9)
entering panic shell loop
! QEMU: Terminated
```

travisg, 2017-08-10:
Thanks for the report, will look into this.

@nvll this is probably something in the arp cache code you wrote, right?

nvll, 2017-08-10:
I would imagine so. I'll take a look.