littlekernel/lk #201: Custom bootloader for Nexus 5(hammerhead)

dhimanchakraborty, 2017-08-26:
Hello, 

I want to flash the bootloader of a nexus 5 to a custom bootloader. The custom bootloader is needed to perform some custom booting cryptographic verification like secure boot/authenticated boot So I have a couple of questions.

1. fastboot flash bootloader <bootloader.img> this command will flash the bootloader or something else?
2. Can I build the little kernel, create an img file and flash the stock bootloader with the custome bootloader?

Thanks,
Dim

travisg, 2017-08-26:
Sadly, no. This littlekernel project is the original project which was forked by Qualcomm and used for bootloaders. This project has no affiliation with Qualcomm or can be used to run on smartphones.

closed 2017-08-26 by travisg

M1cha, 2017-08-27:
@dhimanchakraborty 
1) yes, this will flash the bootloder
2) Flashing any self-built bootloader would brick your phone because Nexus5 has soc security enabled which requires all binaries to be signed with Googles private private key.
3) since your reason for this is security you need a board where you can burn your own security keys. I'm not aware of any OEM or soc manufacturer providing the hardware and/or tools/information to do so. Please correct me if I'm wrong.

dhimanchakraborty, 2017-08-27:
@M1cha Thanks for the answer. 
I am able to get the code that qualcomm uses as bootloader from https://source.codeaurora.org/quic/la/kernel/lk .  
Isn't there anyway to bypass SoC security. 
If I am not wrong if the bootloader is unlocked then it does not look for any signature and an custom update to bootloader is possible. Is it really possible?

Thanks
Dim

M1cha, 2017-08-27:
@dhimanchakraborty Unlocking the bootloader just means that the signature-check of boot.imgs get's disabled. There's no way to disable SoC security(the enable-bit has been burned into hardware).

bzhu5, 2017-08-29:
Right, Unlocking doesn't mean that the bootloader can be unlocked. Some of SoC vendors may provide a solution to unlock SoC security for debugging only purpose (although that bit is fused after production), but only authorized OEM can do that with a secure debug token.  so probably you need a new developer board rather than production Nexus 5.

rahulvermag4, 2018-10-10:
TMC

moe123, 2018-10-11:
Hello, depends on your board generation, some folks managed to access Qualcomm `TrustZone`, even if it won't give you all what you would expect ; you can manage from there by getting the key.

tallero, 2021-09-05:
news on this?

M1cha, 2021-09-05:
@tallero I don't know but there's been a few trustzone exploits over the years - you could try to find out if they can be used.