littlekernel/lk #202: lk can't run on AMD Dell Ryzen7

eastmoutain, 2017-08-29:
hi
     I managed flashing LK kernel and creating a bootable USB device driver, it can boot on any Intel computer, but failed on AMD Dell Ryzen7.  

Here is experiment procedures:
    1.  run `./script/do-qemux86 -6` 
         this will generate lk.elf
    2. create an ISO image
        a) create a directory tree looks like bellow. note '^' stands for space key
           isofile
              |---boot
              |^^^|---grub
              |^^^|^^^|--- 
         b) copy your lk.elf(find it in your lk build directory) into isofile/boot directory, and create a grub.cfg file under grub diretory.
         c) fill grub.cfg with the following lines
             `set timeout=0`
             `set default=0`
             `menuentry "little kernel" {`
             `multiboot /boot/lk.elf`
             `boot`
             `}`
          d) then your isofile directory tree looks like:
             isofile
               |---boot
               |^^^|---grub
               |^^^|^^^|---grub.cfg
               |^^^|---lk.elf
        run  `grub-mkrescue -o lk.iso isofile`, then lk.iso will be created.
    3. create a bootable USB device driver
        run `sudo dd if=isofile.iso of=/dev/sdb`. Assume your USB device appears as "/dev/sdb" under "/dev" directory.
    4. boot computer
        Inert the  bootable USB device into computer's USB slot, and boot the computer in legacy mode(configured in BIOS) from USB device.

On any recently Intel processor Computer, it works fine, but failed to boot up on AMD processor Computer.

I added the following debug info in "arch/x86/64/start.S" to print "OK" and then halt the processor
`movl  $0x2f4b2f4f, 0xb8000`
`hlt`

It should print "OK" on screen if I insert the snippet at the point before enabling Paging, but won't print after paging is enabled.

It's reasonable that after paging is enabled, CPU should fed with virtual address, so I add kernel offset 0XFFFFFFFF80000000 to 0xb8000 in order to print message on screen.
`mov $0XFFFFFFFF8000b8000, %rax`
`movl $0x2f4b2f4f, (%rax)`
`hlt`

It looks perfect and should work now, however it still can't print the massage.

So am not sure what's going on with LK or AMD processor, I have no idea about that!!!

travisg, 2017-08-29:
Ah I'm pretty sure I know what the problem is. I fixed it over in the
magenta fork. AMD machines are more picky about unused bits in the page
tables and I believe the initial 64bit page table that start.S sets up has
some extra unused bits. I'll see about rolling a fix back.

eastmoutain, 2017-08-31:
thank you very much, i fixed the issue after comparing start.S file with magenta.