littlekernel/lk #236: Is there official released version of LK?

Crisscrisis, 2018-11-06:
Hello everyone,

Now I am learning little Kernel and I want to get the official released version of LK.

On BlackDuck Code Center, I have checked that LK has a relesed version on October 21, 2014, and the home page is here. But it seems that here on 'master' branch, the 'releases' is '0'. Thus I want to know whether the 'master' branch is the official released branch, or is there any released version of LK?

Thank you in advance.

travisg, 2018-11-06:
There are no officially releases, it's more of a moving branch that is continually updated. I'm not sure what the 2014 release is all about, but it's not official.

Crisscrisis, 2018-11-06:
@travisg Thank you for your reply! Since this master branch is continually updated and it stays healthy, can this master branch be understood as a released version? Which means that I just need to follow the master branch. Is that enough?

moe123, 2018-11-06:
@Crisscrisis do what you must. There is nothing preventing you to fork, take what's working for you, hence, maintain your own release process according to your needs.

Crisscrisis, 2018-11-07:
@moe123 OK I got it, thank u :)