littlekernel/lk #242: arm_gic: GICv3 support?

lecopzer, 2019-04-10:
Hello, I have a question about ARM GIC:

Do lk need GICv3 support in `dev/interrupt/arm_gic` ?

Or we have to init GICv3 in `ATF` before load lk loaded as S-EL1 OS and not to init GICv3 in lk?

moe123, 2019-04-10:
reference missing:

https://static.docs.arm.com/dai0492/b/GICv3_Software_Overview_Official_Release_B.pdf

lecopzer, 2019-04-10:
Thanks a lot for reference.
But I wonder whether the GICv3 init should be done inside lk or have to depends on ATF

Thanks for reply

moe123, 2019-04-10:
my 2 cents in lk; going to sleep.

travisg, 2019-04-10:
Indeed, GICv3 support is needed the moment it is brought up on a GIC v3 based machine. Can be tested on qemu if told to emulate V3.

GICv3 is pretty close to 2, but different enough that it's worth making a copy of the v2 driver and modifying it.

travisg, 2019-06-27:
We made a gicv3 driver for zircon, so probably simpler to just back port it from there.

gilbert0225, 2024-12-19:
@travisg , Hi, I hope to try back-porting using gicv3 driver of zircon. Can you share proper link of the driver to refer ?

travisg, 2024-12-20:
Sounds good! It shouldn't be too hard, just a case of sorting out the
differences between the two forks of it. I'd really appreciate it!

> Message ID: ***@***.***>
>

travisg, 2024-12-20:
Oh actually I started a branch a while back to do the same thing.
https://github.com/littlekernel/lk/tree/wip/gicv3 looks like it was pulling
the gic stuff back from trusty. I can try to get that finished as well.
Looks like I was just in the debugging phase of the project.

> Message ID: ***@***.***>
>>
>

gilbert0225, 2024-12-21:
tracisg, thanks. I tried to port zircon's gicv3 from your guidance and it seems work. But as I just copy the code, folders and files structure not very clean. I'll try your wip/gicv3 also next week. I hope to contribute gicv3 porting/testing also. :)

travisg, 2024-12-21:
Yah one of the issues I have personally is aside from QEMU I dont have anything handy to test GICv3 on, since all of the dev boards I have floating around are all GICv2 based. I would have added support for it a long time ago otherwise.

gilbert0225, 2024-12-26:
With my test board with Cortex-a55, I executed LK from BL31 (Exit EL3 to normal world).
And Timer seems working with following code with 'time' and 'timeh' command.

```c++
#define PRMS_VIRT_BASE              (0xFFFFFFFF00000000)
#define PRMS_CPU_SPACE_BASE_PHYS    (0x2C000000U)
#define PRMS_CPU_SPACE_SIZE         (0x01000000U)
#define PRMS_CPU_SPACE_BASE_VIRT    (PRMS_VIRT_BASE + PRMS_CPU_SPACE_BASE_PHYS)
#define GICBASE(n) ((uintptr_t)PRMS_CPU_SPACE_BASE_VIRT)
#define GICC_OFFSET (0x300000)
#define GICD_OFFSET (0x400000)
#define GICR_OFFSET (0x460000)

void platform_early_init(void) {
    /* initialize the interrupt controller */
    arm_gic_init();
    arm_generic_timer_init(ARM_IRQ_NOSEC_PHY_TIMER, 1000000);
``` 

```
arm_generic_timer_init:330: register irq 30 on cpu 0
Generic timer initialized with freq 1000000 Hz, irq 30
...
initializing ports
creating bootstrap completion thread
read_cntpct:165: cntpct: 0x000000000590b287, 93368967
timer_set:74: timer 0xffff00000005d688, delay 10, period 10, callback 0xffff00000002c8d0, arg 0
read_cntpct:165: cntpct: 0x000000000591ffdf, 93454303
timer_set:88: scheduled time 93464
insert_timer_in_queue:58: timer 0xffff00000005d688, cpu 0, scheduled 93464, periodic 10
timer_set:99: setting new timer for 10 msecs
write_cntp_tval:157: cntp_tval: 0x00002710, 10000
write_cntp_ctl:147: cntp_ctl: 0x1 0
top of bootstrap2()
``` 

But, when 'sleep 2' command has no response with following log print.
Could you give me some guide what I can try to find the root cause ?

```
falcon] sleep 1
timer_set:74: timer 0xffff000000006c80, delay 1000, period 0, callback 0xffff00000002ce80, arg 0xffff0000000049b8
read_cntpct:165: cntpct: 0x000000001488c158, 344506712
timer_set:88: scheduled time 345507
insert_timer_in_queue:58: timer 0xffff000000006c80, cpu 0, scheduled 345507, periodic 0
read_cntpct:165: cntpct: 0x00000000148a5638, 344610360
read_cntpct:165: cntpct: 0x00000000148ad405, 344642565
read_cntpct:165: cntpct: 0x00000000148b4fad, 344674221
timer_cancel:187: setting new timer to 833
write_cntp_tval:157: cntp_tval: 0x000cb5e8, 833000
write_cntp_ctl:147: cntp_ctl: 0x1 1
```

gilbert0225, 2025-01-02:
@travisg, from my CA55 device test, I added more API and initiation code as below. Can you please review this ?
https://github.com/littlekernel/lk/pull/427

gilbert0225, 2025-01-06:
Right after GICD/GICR address updated from arm_gic_init, the GICD_CTLR has 0x50.
It was in a single security mode, and secure interrupts affinity routing was enabled
#define GICD_CTLR_DS              (1U << 6)
#define GICD_CTLR_ARE_S         (1U << 4)

So, GICV3_IRQ_GROUP (GRP1S 1<<2) mask couldn't applied from EL1 mode.

My test board has 0x12 for GICD_CTLR after BL1/2/31 execution.
So, it's necessary to check whether the enabled GICD_CTLR_DS condition is okay or when it was configured.

I'll check futher.