littlekernel/lk #244: Is it possible to run lk on STM32F401CCU6 ?

zinahe, 2019-05-21:
Hi,

Sorry if this question has been already been addressed before. I would like to tinker with lk and I'm wondering if I can run it on a board with STM32F401CCU6 (256K Flash/64K RAM).

Thanks,

Zinahe A.

swetland, 2019-05-21:
Looks like it mostly has been used with 32F407/417 previously (0f the F4xx sub-family) so you'll probably need to tinker with things slightly for 401 (based on a quick peek at https://github.com/littlekernel/lk/blob/master/platform/stm32f4xx/rules.mk), but it certainly should be doable.

travisg, 2019-05-21:
Indeed, as swetland points out it should be runnable. It's been a while since I looked at the 400 family, but I think the 401 and 407 are fairly close. It may require an update of the STM software drop that is in the tree but those are usually quite configurable.

zinahe, 2019-05-21:
Great! Thanks for the quick reply. I bought these dirt-cheap STM32F401 boards on [AliExpress ](https://www.aliexpress.com/item/33011237530.html?spm=a2g0s.9042311.0.0.61624c4dPco7ux) for less than $5 to play with. I'll report my findings here.

travisg, 2019-05-21:
Oh that's a nice board. I should grab one too.

zinahe, 2019-06-25:
@travisg,
It took a little while but I was finally able to run lk on the tiny STM32F401 board. Most of my time was actually spent in trying understand the internals of the build system; which I have to say is very well organized. I'd be happy submit a pull-request if there is any interest.
Cheers,

travisg, 2019-06-27:
Sure, I'll roll in the commit. Thanks!

zinahe, 2019-06-27:
Thanks. Pull [request ](https://github.com/littlekernel/lk/pull/245)is submitted. Let me know if there is any issue.