littlekernel/lk #247: Why is "WITH_LIB_CONSOLE" not defined anywhere?

zinahe, 2019-07-01:
Hi,

I have done a full scan of the code trying to understand **WITH_LIB_CONSOLE**. I have have seen many cases where it is being tested for conditional compilation of blocks of code. But it doesn't seem to be defined anywhere. Not in any header file; nor in any make file; not even in **app/shell** which seems to me the primary user of **lib/console**. What gives ?

Thanks,

travisg, 2019-07-01:
The WITH_* #defines are set by the build system when that particular module
is included in the build. It's not used a lot, but WITH_LIB_CONSOLE is
tested in lots of places to determine if it should add a shell function.

travisg, 2019-07-01:
If you look in build-<project>/config.h it's set there, the config.h
file(s) are generated by the build system.

zinahe, 2019-07-01:
@travisg,

Thanks for the quick response. So, it's auto-generated at build time. That explains why I was never able to pinpoint its define in the source tree. Basically, I'm trying find ways to generate the smallest possible binary for a blinky app on an STM32F401 board. One that seemed help so far is **GLOBAL_COMPILEFLAGS +=  -DDISABLE_DEBUG_OUTPUT=1** in the project/<name>.mk file. That shaved off a whopping 8k from the bin file. I'd be grateful for any pointers.

travisg, 2019-07-01:
> += -DDISABLE_DEBUG_OUTPUT=1* in the project/.mk file. That shaved off a

Ah a project with just an app that does the blinky, and the
DISABLE_DEBUG_OUTPUT should be pretty close. Also, set it as
GLOBAL_DEFINES += DISABLE_DEBUG_OUTPUT=1. That gets more cleanly added to
the config.h files, though it should end up with about the same thing.

Removing debug output kills printf, which generally takes out a sizeable
chunk of stuff. IIRC, the smallest build with the kernel is around 10K or
so. The ST libraries are not great though and tend to add a fair amount of
bloat, though they usually compile out if not called.

travisg, 2019-07-01:
Oh, set debug level to 0 as well, that removes almost all of the asserts in
the system. You can do it with DEBUG:=0 in the project file.