littlekernel/lk #249: is it that hard to upstream platform support?

Mis012, 2019-07-18:
Hi,
though the MIT license seems to hint at little care for end users, I find it really sad that even though lk is used on so many platforms, the upstream codebase supports hardly any of them in any way.
Has the focus of lk changed to not include the android bootloader etc use case, or do you not mind that the "Who is using LK?" section on your wiki doesn't include a single achievement of the upstream codebase, and no one can actually use upstream LK for any of those things because it doesn't support almost any platforms other than MCUs?
I understand that e.g qcom doesn't seem to have any interest in upstreaming their SoC support to lk, but what is actually the focus of lk if all those downstream forks which seem to be the most visible achievement of lk will never see any benefit from it's continued development?
Sorry if there's not a constructive way to solve this issue, but it makes me very unhappy

nvll, 2019-07-18:
The permissive licensing allows for various downstream users to use the project, but corporations like Qualcomm have little incentive to upstream their changes, especially if it means coordinating patches with LK contributors. LK itself is still used by folks for personal and educational projects and we don't have any way to list those.

Even if changes aren't upstreamed LK is still a major part of bootloaders, embedded projects, and trustzone work for companies like Qualcomm, NVIDIA, Google, etc.

moe123, 2019-07-19:
@Mis012 what you ask is a culture of ethics; unfortunately, this has been long forgotten. When people feel untitled to do whatever they want because of anything (the list of reasons is by definition indefinite).

Mis012, 2019-07-22:
I wonder, if you went GPL for new code, would they still care then?  
Since they don't really care about the upstream... And I assume people using the upstream codebase would not mind? Not that it would solve the platform support issue, but I know I would not care about adding that support myself if I can't upstream it under a copyleft license. Not that I think I would be necessairly capable of it.  
Maybe it's just that few hobbyists care about the parts of these SoCs that are not linux, more so when you need to chainload the lk because of *cough* signature verification?  
But even with chainloading, *this* is the proper way to enable stuff like booting linux from a filesystem on phones, or simple kernel chooser menu that comes with that. I can manually cherrypick stuff to do that in ugly qcom downstream, but even getting it to work on more than one SoC with the same codebase may be impossible...  
I hate the idea that all of my work would need to be duplicated to achieve this on another device, and the small stuff that may still work with upstream and could be upstreamed  I'd have to give out under a non-copyleft license, contrary to may personal beliefs.
That's also an issue with non-copyleft upstream, as with copyleft upstream, you can technically have 95% of the code non-copyleft, so people from neither camp need to go against their beliefs in order to upstream their changes.
That being said, if someone goes as far as reimplementing GPL code *purely* to make it non-copyleft, I guess that's where the line is for me in viewing them as like-minded.

@moe123 the assumptions that ethics cannot be counted upon is why GPL is a thing. That being said, forcing everyone to upstream changes as well is not really an option, so I'm not sure there's much that can be done in that regard

moe123, 2019-07-22:
@Mis012 Hello, I don't GPL has something to do with upstreaming; which is a mindset; ethic; something legal tools won't ever cover. For instance, look at the different forks of the linux kernel in the embedded world; same for gcc forks and rebranded toolchains; same for Android they never upstreamed anything or even tried to generalized ideas; "disputing" their "choices"; in my opinion, it would have helped them not to fall in many conceptual traps; however that's another topic.

Mis012, 2019-07-22:
I don't think there's a way to force someone to upstream. But Linux is a) lovely and b) GPL, and there are actually people mainlining support for devices that would otherwise be left with ugly downstream mess. Not much SoC upstreaming I guess, but it's theoretically doable.
I can't imagine putting that amount of work towards a non-copyleft project, and I doubt anyone but the vendor would (and the vendor clearly doesn't care about upstreaming)

I guess that putting that much work towards a project you use just as a bootloader may be a part of it, but some things just belong in the bootloader.

moe123, 2019-07-23:
@Mis012 Hello, I don't believe GPL has fulfilled is intend; because GPL enforces a dominium;  people would prefer participation than enforcing to comply to any ideology; that's the main problem behind GPL; for people who don't find the absolute need in their life to have a political opinion or a say or be part of any activism. However I like ethic and you won't find it in any ideological constructs. "I can't imagine putting that amount of work towards a non-copyleft project"; this is where people differ, people don't forcibly seek glory or reward; pure and simple intellectual interest is sufficient to them.

Mis012, 2019-07-23:
It's not about glory or reward, GPL is trying to save the hacker culture, and Linux is a massive success. Though that's harmed by tivoization, many devices are hackable thanks to running GPL'd Linux.  
MIT says "do whatever you want"  
GPL says "do whatever you want, no matter how far in the chain you are"  
I believe LGPL is more/ fair for libraries, because how the rest of the program is licensed is not my business.
If qcom was a bit less nice, I would not be able to run my own lk at all...

moe123, 2019-07-23:
@Mis012 hello nope because you have a partial use of it: GPL says what you do belongs to GPL;  this is a dominium system, or alienation system; you must comply or else! "GPL is trying to save the hacker culture" in the fact, it killed it; GPL'd Linux is a lucrative business with a set of ideological dystopia to fool the fools.

closed 2020-05-17 by travisg