littlekernel/lk #250: What's up with "-finline" in engine.mk ?

zinahe, 2019-07-23:
Hi,

This could be due to lack of knowledge; so be gentle. I have not been able to find an **-finline** option in GCC. One that comes closest to it is **-finline-functions**. 

Grateful if someone could shed some light on this.

moe123, 2019-07-23:
Hello,

perform c++ inlining.

travisg, 2019-07-23:
Yah, it goes pretty far back, and it may be unnecessary with modern gccs,
but the main idea is to try to force the compiler to honor inline, even if
being built with -O0.

It may be mostly unnecessary now that inline is part of both C and C++, but
in the older days you really needed to switch it on or it might not do it.

zinahe, 2019-07-24:
Thank you for the prompt reply. Why not remove it then ? It's not like you'd be breaking any backwards compatibility. 

Food for thought.

nvll, 2019-07-24:
If it's not hurting anything it's fine to leave it there rather than risking unexpected regressions by removing it.

moe123, 2019-07-24:
@hello, I think my comment has been misinterpreted;

1- you must keep it; else when it's legit it won't be honored.
2- let's write a binomial coefficient recursive function for integers

```c
inline unsigned int nchoosek(unsigned int n, unsigned int k)
{
	if (n < UINT_MAX && k < UINT_MAX) {
		if (k > n) {
			return UINT_MAX;
		} else if ((k == 0) || (k == n)) {
			return 1;
		} else if ((k == 1) || (k == n - 1)) {
			return n;
		}
		return (n * nchoosek(n - 1, k - 1)) / k;
	}
	return UINT_MAX;
}
```

conclusion:  in this case, we all do understand why the inlining request won't be honored.

You have a good day.

travisg, 2019-07-25:
Really, I'd just need to be convinced it's not needed on modern toolchains
(say 4.x gcc series) for both C and C++. If so, then it'd be worth
dropping. Otherwise it's probably harmless.

moe123, 2019-07-25:
@hello use `-Winline` and you'll know.

zinahe, 2019-07-26:
Again; apologies if I didn't make my question clear. I was not asking if `-finline-functions` is useful or not in general. I was interested to know why it is written as `-finline` in engine.mk. I had no idea whether it a short-hand for `-finline-functions` or it means something completely different ?

@moe123,
Anyway, in an effort to answer my own question, I compiled lk with `-Winline`. I got the exact same  binary with `-finline`, with `-finline-functions` and with both removed. GCC reported the same number of warnings failing to inline some functions (148 to be exact).

I'm not sure what it all means. But if one gets the same result with or without `-finline` then what is it doing ?

moe123, 2019-07-26:
@hello, it depends on your optimization flags see documentation for property inheritance; + don't mixup c and c++ optimization flags.

zinahe, 2019-07-26:
Setting ARCH_OPTFLAGS to -O2 and -O3 made no difference.

swetland, 2019-07-26:
I believe the situation that lead to the use of -finline originally was that under -O0 (and possibly -Os?) the compiler would not honor the inline keyword (as Travis mentions upthread), leading to compilation errors at those optimization levels.

doug65536, 2020-04-04:
If you want to force inline in -O0, use `inline __attribute__((__always_inline__))`

travisg, 2020-05-17:
I finally removed the -finline switch. A bit late to get to it, but it was indeed extraneous.

Looks like it literally does nothing *except* to cancel an existing -fno-inline switch on the command line. GCC internally tracks it literally as flag_no_inline and defaults to 0.

closed 2020-05-17 by travisg