littlekernel/lk #254: aarch64 backtrace not appearing on exception

vharron, 2019-09-30:
REPRO:

Host: x86_64 Debian stable

Linaro aarch64 toolchain: gcc-arm-8.3-2019.03-x86_64-aarch64-elf

Qemu built from recent ToT:
commit 240ab11fb72049d6373cbbec8d788f8e411a00bc

cd ~/eh/model/ && \
git clone https://git.qemu.org/git/qemu.git && \
cd qemu && \
git submodule init && \
git submodule update --recursive && \
./configure --enable-debug-info && \
make -j$(nproc)

LK commit 9a6081399cd6b57e5bc7504511e160e33c3cf488

With this patch:

diff --git a/lib/console/console.c b/lib/console/console.c
index efa4262b..a2bbd075 100644
--- a/lib/console/console.c
+++ b/lib/console/console.c
@@ -693,6 +693,7 @@ void console_abort_script(void) {
 void console_start(void) {
     debug_buffer = malloc(LINE_LEN);
 
+    *((volatile uint16_t*)0xcafe) = 1138;  // Force exception.
     dprintf(INFO, "entering main console loop\n");
 
 
$ scripts/do-qemuarm -6

OBSERVED:
qemu-system-aarch64 -machine virt -cpu cortex-a53 -m 512 -smp 1 -kernel build-qemu-virt-arm64-test/lk.elf -nographic

welcome to lk/MP

boot args 0x0 0x0 0x0 0x0
INIT: cpu 0, calling hook 0xffff000000031060 (version) at level 0x3ffff, flags 0x1
version:
	arch:     arm64
	platform: qemu-virt-arm
	target:   qemu-virt-arm
	project:  qemu-virt-arm64-test
	buildid:  I9UIB_LOCAL
INIT: cpu 0, calling hook 0xffff000000032a58 (vm_preheap) at level 0x3ffff, flags 0x1
initializing heap
calling constructors
INIT: cpu 0, calling hook 0xffff000000032ab0 (vm) at level 0x50000, flags 0x1
initializing mp
initializing threads
initializing timers
initializing ports
creating bootstrap completion thread
top of bootstrap2()
INIT: cpu 0, calling hook 0xffff00000002e5e8 (pktbuf) at level 0x70000, flags 0x1
pktbuf: creating 256 pktbuf entries of size 1536 (total 393216)
INIT: cpu 0, calling hook 0xffff0000000311c8 (virtio) at level 0x70000, flags 0x1
creating bootstrap completion thread for cpu 1
creating bootstrap completion thread for cpu 2
creating bootstrap completion thread for cpu 3
initializing platform
initializing target
calling apps_init()
starting app inetsrv
starting internet servers
starting app shell
data fault: PC at 0xffff000000037844, FAR 0xcafe, iss 0x44
ESR 0x96000044: ec 0x25, il 0x2000000, iss 0x44
iframe 0xffff0000001561d0:
x0  0xffff000000156340 x1  0x            cafe x2  0x             472 x3  0xffff0000001563c0
x4  0x              27 x5  0xffff000000154f50 x6  0xffff000000086508 x7  0x               0
x8  0x               1 x9  0xffff00000003c000 x10 0xffff00000014c400 x11 0xffffff80ffffffc8
x12 0xffff00000014c440 x13 0xffff00000014c440 x14 0xffff00000008b000 x15 0xffff00000014c360
x16 0x               1 x17 0x               0 x18 0x               0 x19 0xffff000000154f50
x20 0xffff000000086478 x21 0x               0 x22 0x               0 x23 0x               0
x24 0x               0 x25 0x               0 x26 0x               0 x27 0x               0
x28 0x               0 x29 0xffff0000001562e0 lr  0xffff000000037834 usp 0x               0
elr 0xffff000000037844
spsr 0x        60000305
panic (caller 0xffff00000001125c): die
CRASH: starting debug shell... (reason = 9)
entering panic shell loop
!

EXPECTED:
backtrace including 

ADDITIONAL INFORMATION:
If this is an RTFM moment, I would really appreciate a pointer to the manual.  Thanks!

vishals4gh, 2020-07-22:
https://github.com/littlekernel/lk/pull/267/commits/ae403964787d2df688787d6e37d08ce6160249a8 might help here.

travisg, 2020-11-27:
@vishals4gh 's change made it in, so going to close this one out for the time being.

closed 2020-11-27 by travisg