littlekernel/lk #262: Question about the fat32_mount function and the management of dynamically allocated memory

tec-quark, 2020-05-14:
Hi, sorry for my English, I am using an online translator, I have a doubt about memory management in the fat.c file precisely in the fat32_mount function. I would like to know what happens with the memory allocated to the fat_fs_t * fat pointer in the event of an error. How is memory freed?

```

    fat_fs_t * fat = malloc (sizeof (fat_fs_t));
    fat-> lba_start = 1024;
    fat-> dev = dev;

    fat-> bytes_per_sector = fat_read16 (bs, 0xb);
    if ((fat-> bytes_per_sector! = 0x200) && (fat-> bytes_per_sector! = 0x400) && (fat-> bytes_per_sector! = 0x800)) {
        printf ("unsupported sector size (% x) \ n", fat-> bytes_per_sector);
        result = ERR_NOT_VALID;
        goto end;

```

Thank you very much
Santiago

moe123, 2020-05-15:
fat is assigned to cookie in case of error or not it will be freed in fat32_unmount.

tec-quark, 2020-05-15:
Thank you for your answer and suggestion. I'd like to try and emulate the error to see what happens, but I preferred to ask this question here to get an opinion from people who have much more experience than me. I will do it. Also I would like to add some code to check the correct memory allocation by making a simple `if (fat == null)`, for security.
Thanks again for your suggestion. A greeting

moe123, 2020-05-15:
you must set cookie to null else you'll hit dandling condition; there is no implicit state/assignement in C.

tec-quark, 2020-05-15:
Yes, sorry, I misunderstood myself, I was also referring to this section of the code
```
    fat_fs_t * fat = malloc (sizeof (fat_fs_t));
// here
    fat-> lba_start = 1024;
    fat-> dev = dev;
```

moe123, 2020-05-15:
yes I know; I went further more; I'll always do; if you are able to grasp the first step I'll always imply next.

travisg, 2020-05-16:
Indeed it looks like that code is in some need of cleanup for error cases. Any patches are welcome to help clean it up a bit.

tec-quark, 2020-05-16:
Hi, yes, I was encouraged by moe123 and I have already forked the project (which I find fantastic) and I am making some changes on the code. Then I will try to understand how to make the code visible, this is the first time that I work on an external project :) Thanks.

if my english is not good ... then it's the fault of the online translator :)

moe123, 2020-05-16:
@tec-quark  your translator is doing fine; better than I do sometimes. However, if you
have a task within a source base, first don't let yourself distracted go for it.

Few informal advices: when changing a code there are two steps:

- 1 The code itself *
- 2 Its contextual usage (might drive choices in #[1])

Meanwhile, in this particular situation there is a third one: fat is part of a larger family. Does
all members of this family are homogeneous and comply to the same rulesets at usage.

[*] you may perfectly emulate it on your desktop as long as it is a unix-like, devices can be seen as plain file.