littlekernel/lk #275: [feature] refresh optimized ARM string routines

travisg, 2020-10-11:
Pull new copies of the optimized arm routines from https://github.com/ARM-software/optimized-routines

Alvarito050506, 2021-05-30:
Hello! I would like to work on this, but I'm not sure if the updated versions should use the VFP and/or NEON, or not. Maybe using these features could be platform-dependent?

travisg, 2021-05-30:
Oh sure, that'd be lovely. I'd start with the non neon ones yes. For better or worse, most of the LK code runs in supervisor mode and some deep kernel code may call into the same routines and wouldn't be capable of using fpu/neon. So, in general most of the core library routines cant touch the fpu.

travisg, 2021-05-31:
As a note, since they're coming from an external source, put the routines in the external/ top level directory and then have the lib/libc/string/... make logic just reach over and build the stuff out of there. That makes it easier to independently track the external repository, for future updates.