littlekernel/lk #278: Not able to receive interrupts in little kernel

alekarasma, 2020-10-24:
Hello Everyone,

We are using Trusty in our project which uses little kernel for security in TrustZone, our microcontroller is IMX7D from NXP. I have added GPIO and SPI driver in the Little kernel for imx7 platform and configured GPIO for rising edge interrupt. However, I am unable to get interrupts running in the Little kernel. If I delete the interrupt configuration from little kernel I am getting interrupts in Linux kernel with no problem. Is there any configuration needed at Linux side or Little kernel side to pass interrupts from Linux to the Little kernel?

Any help is really appreciated.

Thanks,
Asma

travisg, 2020-11-27:
I honestly don't know, you'll have to ask the Trusty folks about that one. They're using a fork of LK and I haven't been following along in some time. It's diverged enough that they're probably going to need to look at it.

alekarasma, 2020-11-28:
Thanks Travis for the reply,

We didn't get interrupts directly working in Trusty, but managed to forward interrupts from Linux to the Trusty through the secure monitor.

Asma

closed 2020-11-28 by alekarasma