littlekernel/lk #295: could not set breakpoint on _start or arm_reset

wooshifu, 2021-03-30:
**project**: rpi3-test
**note**: build with macro `WITH_KERNEL_VM=1`
**issue**: could not set breakpoint on _start or arm_reset
**steps:**
1. run `qemu-system-aarch64 -s -S -M raspi3 -kernel lk.elf -semihosting -serial null -serial mon:stdio -nographic` to start the debug session
2. run `aarch64-elf-gdb lk.elf` to start debugging
3. `target remote :1234`
3. `b _start`
4. `b arm_reset`
5. `si` shows `0x0000000000080004 in ?? ()`
6. `disassemble arm_reset` shows `Cannot access memory at address 0xffff000000080000`
7. `b lk_main`
8. `c` then gdb can break at `lk_main` as extecpted

**questiones:**
- why gdb says `Cannot access memory at address 0xffff000000080000`?
- why gdb command `b _start` or `b arm_reset` not working?
- why gdb command `b lk_main` works?

[image]

travisg, 2021-08-08:
The problem here is the cpu hasn't turned on the MMU at this point, and the address you're trying to set the breakpoint on is the virtual address once the mmu is enabled. The first bit of start.S is running with mmu disabled, so the address the cpu is running at is a physical address.

If you set the breakpoint a bit farther on, once the mmu is enabled it probably will have a better chance of working.

wooshifu, 2021-09-11:
got it. thanks a lot😁

closed 2021-09-11 by wooshifu