littlekernel/lk #301: Question about relation between VM and SMP

BY1994, 2021-08-04:
Hi, I want to ask a question about SMP config.

The config named "WITH_SMP" seems that it depends on the config named "WITH_KERNEL_VM".

arch/arm64/start.S
`#if WITH_SMP
    cbnz    cpuid, .Lsecondary_boot
#endif
#endif /* WITH_KERNEL_VM */
`
Those two configs are dependent?

travisg, 2021-08-08:
Ah, yes that could be refactored. I in general the two are separate concepts, but in the current code base the arm64 code mostly runs with mmu enabled because there hasn't been a need to do otherwise. I think some careful restructuring there should work.

The harder problem is having the secondary cores run with the mmu off while the main one runs with it on, since basically all of the code in the system assumes it's relocated to a particular virtual address. If what you're asking for is for some simple code that runs semi independently, then perhaps a better strategy is to define an alternate secondary cpu entry point that just sets up enough and doesn't try to enter the rest of the kernel.

BY1994, 2021-08-10:
Thank you for the kind reply!! I have used lk as bootloader with mmu off and SMP off so far. And now I'm trying to use SMP for fast booting by using threads. So I tested the codes with SMP on and mmu off, and it worked well except spin_lock. Although it worked by adding sev instruction in spin_unlock procedure (because an event is not triggered automatically when cache is off, I added it manually), it would be safe and reasonable to use virtual memory if the codes assume relocated virtual address as you told. It might cause problem in some point. Many thanks again!

closed 2021-09-05 by BY1994