littlekernel/lk #330: newlib - unknown type name 'clock_t'

Inchul-Lee, 2022-06-23:
I am using little-kernel with newlib or newlib-nano for using the time functions. But I am facing with error as the following.
[image]
Is it a issue or my mistake?

travisg, 2022-06-23:
LK doesn't define clock_t anywhere inside the tree, but it has it's own notion of time that may sync up with clock_t.

See lk_time_t or lk_bigtime_t. It may be that what newlib is expecting clock_t to be would line up with that, and  you could locally typedef clock_t as lk_time_t

Inchul-Lee, 2022-06-24:
Thank you for your relpy. Actaully I need functions like as localtime() ans so on. How can I use this functions? For your reference I attached a part of my code.
[image]

travisg, 2022-06-24:
You'll have to implement it, since LK has no concept of localtime() and doesn't track any sort of notion of time zones or time of day. It shouldn't be *too* difficult to implement, but there will have to be some notion of synchronizing time with an RTC and then tracking localtime.

Inchul-Lee, 2022-06-24:
OK, So I am trying to apply newlib for mktime() and localtime() functions. If you have a idea to apply newlib into little-kernel, please give me some information.

Inchul-Lee, 2022-06-24:
I decided to implement the function for time according to your guide.
Thank you.

closed 2022-06-24 by Inchul-Lee

travisg, 2022-06-25:
Yah, I've thought about newlib from time to time. The general issue is for everyone that wants to use newlib, someone wants to use another libc, so in general LK doesn't take a stance on it. However, it does mean these sort of things show up and there's not a good single solution except to try to support *all* of the libcs and have a compile time switch. I think it'd be technically doable, but it'd be a huge amount of work.

Happy to see someone using LK! Is this for a personal project or something else?

Inchul-Lee, 2022-06-27:
Hi, This is my company's project. I think that the newlib is good solution. But the flash size of my producut is  limited. If I use newlib, I assume that the flash memory size is not enough. Thank you for your attention.