littlekernel/lk #361: Rust

Surysingh, 2022-11-08:
Rust has lot of features which can help prevent common memory related issues.  Recently Linux Kernel is also moving towards Rust.

Is there any plan for LK to also use Rust?

Thank you.

travisg, 2023-01-16:
I've spent some time this last holiday learning some rust and it is definitely something that could be done. The hard question is defining exactly how to fully integrate it into the system. ie, which direction can apis call? Is it okay to define a rust module that C code calls into, or is it more of a C -> Rust as some sort of leaf node library.

Surysingh, 2023-01-25:
> code

Than you for the reply.

For example,  the memory manager like dlmalloc written in Rust,  the malloc-free api users are written in say C/C++,  then the benefits offered by rust,  will no longer apply in C program.  So would it be very effective,  is this the question being considered?

kanakshilledar, 2024-10-06:
Are there any plans of moving towards rust?