littlekernel/lk #362: The lk coding style question

ccteam, 2022-12-08:
Dear lk contributors and reviewers:
   I have a simple question: lots of projects use different coding style. For lk history, i found that it uses tab for indent for some time, and some years ago it was changed to "4 spaces indent". Why?

regards,
xi

travisg, 2022-12-08:
For the most part it's because I kept getting lots of grief from lots of folks about using tab indents since they weren't in style at the time, and so I switch to 4 space indents since that was my preference. Since then I've generally adopted 4 space indents for all my projects since I've grown to like it.

Individual code is free to use whatever it wants, but to be consistent i like to keep the main LK repo in at least one indent style so it doesn't get too messy.

ccteam, 2022-12-09:
Thanks. We found a question that if using "4 spaces indent", sometimes, if i wrote a wrong indent, wanted to delete the indent(4 spaces). It needs four DELETE key. A better way for this?

travisg, 2023-01-16:
Some code editors support deleting tabs like this, but personally I've just found it to be second nature to delete 4 spaces when using editors that don't have that intelligence. But it is an interesting question: what editors do?

closed 2025-01-10 by travisg