littlekernel/lk #363: [Feature] Dual boot support

GreenLunar, 2022-12-21:
Add a feature to choose which install to boot each time a device turns on or allowing some kind of configuration for changing which install is preferred on boot.

Source https://github.com/msm8916-mainline/lk2nd/issues/100

luka177, 2022-12-22:
are you kidding? Even single-booting is out of the scope of this project, this project is just kernel, not bootloader. Qcom platforms did use OLD and MODED lk to boot linux + even for it there is dualboot solution, check https://github.com/Android-Boot-Manager

luka177, 2022-12-22:
Issue should be closed now

GreenLunar, 2022-12-22:
Thank you for the reference.

I'll look into it.

P.S. I'm attempting to dual boot multiple pure Linux OSs.

travisg, 2022-12-22:
Unfortunately luka177 is right. The Qualcomm bootloader you're probably referring to is a really really old fork of LK (like 2009 era) that has diverged from mainline in a ton of ways. I honestly don't know what it's doing with any of these things, so you'd really have to start with that codebase and work from there.

closed 2022-12-22 by travisg