littlekernel/lk #387: Evaluate Profile-Guided Optimization (PGO)

zamazan4ik, 2023-10-12:
Hi!

Recently I did many Profile-Guided Optimization (PGO) benchmarks on multiple projects - the results are available [here](https://github.com/zamazan4ik/awesome-pgo). Here you can find different applications from different domains that were accelerated with PGO: operating systems (like Linux and Windows kernels), virtual machines (like QEMU and CrosVM), compilers, gRPC workloads, benchmark tools, databases, and much more. So that's why I think it's worth trying to apply PGO to LK.

I can suggest the following things to do:
* Evaluate PGO's applicability to LK via benchmarks.
* If PGO helps to achieve better performance - add a note to LK's documentation about that. In this case, users and maintainers will be aware of another optimization opportunity for LK.
* Provide PGO integration into the build scripts. It can help users and maintainers easily apply PGO for their own workloads.

After PGO, I can suggest evaluating [LLVM BOLT](https://github.com/llvm/llvm-project/blob/main/bolt/README.md) as an additional optimization step after PGO.

travisg, 2024-02-27:
So the question here is how does the system get the data out of the run and back into the build system? In the case of an operating system kernel, the trouble is the data ends up on the device and it's hard to extract it. Is there any prior art to running this over a kernel?

zamazan4ik, 2024-02-27:
> So the question here is how does the system get the data out of the run and back into the build system?

In general case, PGO profiles are saved to a disk. Then, on the optimization phase, these profiles are passed to a compiler via compiler options like `-fprofile-use=pgo_profile_name` (more details can be found in the documentation for your compiler).

> In the case of an operating system kernel, the trouble is the data ends up on the device and it's hard to extract it. Is there any prior art to running this over a kernel?

Yes, there are multiple examples:

* [Linux kernel](https://kernel.org/):
  - [Paper](https://web.eecs.umich.edu/~takh/papers/ugur-one-profile-fits-all-osr-2022.pdf)
  - [Microsoft presentation](https://lpc.events/event/7/contributions/771/attachments/630/1193/Exploring_Profile_Guided_Optimization_of_the_Linux_Kernel.pdf)
  - [ASOS (Application Specific Operating System)](http://scis.scichina.com/en/2018/092102.pdf)
  - [TCP Stream perf](https://lpc.events/event/7/contributions/798/attachments/661/1214/LTO_PGO_and_AutoFDO_-_Plumbers_2020_-_Tolvanen_Wendling_Desaulniers.pdf)
  - [Phoronix post](https://www.phoronix.com/news/Clang-PGO-For-Linux-Next)
  - Yet another attempt to PGO Linux kernel: http://coolypf.com/kpgo.htm
  - [Gentoo Wiki](https://wiki.gentoo.org/wiki/Kernel/Optimization#Performance)
  - Optimizing Linux kernel with Clang. An [article](https://habr.com/ru/companies/ruvds/articles/696236/)(in Russian) and [results](https://github.com/h0tc0d3/linux_pgo)
* Windows: 5-20% improvement according to the [presentation](https://lpc.events/event/7/contributions/771/attachments/630/1193/Exploring_Profile_Guided_Optimization_of_the_Linux_Kernel.pdf)

E.g. in the Linux kernel case, profiles can be gathered via `procfs` (AFAIK) and then passed to the compiler on the optimization phase.