littlekernel/lk #398: [virtio/fs] Support for virtual 9p devices

mob5566, 2023-12-17:
Hi there,

I've implemented a lightweight [virtual 9p device driver](https://wiki.qemu.org/Documentation/9psetup) and file system driver based on `dev/virtio` and `lib/fs`.

I am wondering if it is worth to be merged to the upstream.

If so, is there anyone who can help review those changes?

mob5566, 2023-12-19:
Hi @travisg,

Do you have any comments on this?

Thank you for your time and consideration.

travisg, 2023-12-20:
Seems like a great idea, and could definitely use it. If it's ready to go I'd be happy to look at it in a pull request.

mob5566, 2023-12-26:
> Seems like a great idea, and could definitely use it. If it's ready to go I'd be happy to look at it in a pull request.

Thanks for the prompt response during holiday season.

It's good to hear that. I need some time to prepare the changes.

I'm closing this issue. I will create pull requests once the changes are ready!

Thanks! Wish you and your family a Happy New Year!

closed 2023-12-26 by mob5566