littlekernel/lk #400: Query regading the Translation

phiyanshu, 2024-02-02:
Hi ALL,
Does the arch_mmu_map function in mmu.c supports mapping for VA>39 bits? 
From what I could observe the MMU_USER_SIZE_SHIFT is dependent on MMU_LX_X which can take maximum value of 39. But in the case of translation of VA>39 bits we need an extra table. Correct me If I'm wrong.

travisg, 2024-02-03:
Hi! Which architecture are you asking about in this case? I'm guessing this is riscv64, since from what I can remember that's the only one that generally defaults to Sv39.

In the case of a Sv48 arch, the code has the logic to deal with it, it just has to be compile time set to a Sv48. See https://github.com/littlekernel/lk/blob/master/arch/riscv/rules.mk#L50

phiyanshu, 2024-02-03:
Hi,
Sorry I forgot to mention the architecture. Thanks for the info.
Another query regarding arm64.
As you can see here:
[https://github.com/littlekernel/lk/blob/master/arch/arm64/rules.mk#L62C1-L62C20]
I was just wondering the USER_ASPACE_BASE will vary according to the user, right? If the user changes it, Will the USER_ASPACE_SIZE will still be the same? I understand the USER APSACE SIZE can go till max 0XFFFFFFFFFFFF.
From what I could understand from the logic to work is that the sum of USER_ASPACE_BASE + USER_ASPACE_SIZE should be 0xFFFFFF000000. Is that the correct understanding or I'm missing something?

travisg, 2024-02-11:
ARM64 indeed supports more than 39 bits. In the default configuration each half of the total address space is 48 bits long, so what those constants are marking is:
user aspace 0x100.0000 - 0x0000.ffff.ffff.ffff
kernel aspace 0xffff.0000.0000.0000 .... 0xffff.ffff.ffff.ffff

The reason for user aspace not being based at 0 is to keep 0 unmapped, so that null pointers are trapped. Actual hardware supports down to 0.

Of course, the *full* 64bit address space is not supported on ARM64 (or most architectures for that matter), so there's a gap of unusable hardware space between 0x0000.ffff.ffff.ffff - 0xffff.0000.0000.0000.