littlekernel/lk #405: Upstreaming msm8916 SoC support

samcday, 2024-05-02:
Hi there @travisg . As I'm sure you're aware, the [lk2nd](https://github.com/msm8916-mainline/lk2nd/) project is an actively maintained LK derivative that was spun out from the CodeAurora (CAF) code. It's used as a chained (from the vendor LK) bootloader in the `boot` partition on a range of msm8916+msm8974+msm8226 devices.

The CAF codebase was forked from LK a very long time ago. Consequently lk2nd looks "kind-of" like LK but [as you noted](https://github.com/littlekernel/lk/issues/363#issuecomment-1363404510) it's diverged quite a lot.

As an educational exercise I recently started exploring porting that downstream work on top of modern LK upstream.

I've managed to get some basic clock + UART working on an msm8196 device in this commit: https://github.com/littlekernel/lk/commit/d21343738bb9d4f59da245eb17dfd605fd100320

To be clear, I'm not asking you to review that code at all yet, since it's 5000 lines of mostly copy/pasta :sweat_smile:

What I'm asking is if you'd be interested in seeing support for these qcom devices brought upstream? If so, I'd be happy to work with you on this, and massage the downstream code for style/quality/nits/etc to make that happen.

The most obvious issue I see is that the downstream code that came from the CAF repo is BSD-licensed, whereas LK upstream is purely MIT. I have no idea how to navigate such a thing. I've always regarded BSD as a very permissive license though. Interestingly, it seems that it might be possible to simply re-license [the BSD code as MIT](https://opensource.stackexchange.com/a/10687), as long as the original copyright notices are retained? I dunno. I am a computer person not a lawyer ;)

omasanori, 2024-08-27:
It would be great if useful part of CAF lk is merged to the mainline.

> Interestingly, it seems that it might be possible to simply re-license the BSD code as MIT, as long as the original copyright notices are retained?

It is safer not to do such thing without permission by the original author.

samcday, 2024-08-27:
> It would be great if useful part of CAF lk is merged to the mainline.

I wonder what constitutes "useful"? :)

I have some [LK out-of-tree code](https://github.com/samcday/lk-msm8916/) with basic working timers/interrupts/UART. I still hope to continue hacking on that some time in the future.

> It is safer not to do such thing without permission by the original author.

Yes, I suspect you're right.

Relatedly, the downstream CAF LK code is ... well, it's a lot like the downstream Android kernel forks of old - stuff is just hacked in all over the place in inappropriate places.

So, ultimately I think the "can we relicense or adopt this code?" question is somewhat moot because the code probably just needs to be mostly rewritten. That's basically what I've been doing in that linked repo, though I haven't been cautious about IP contamination. I assume that a permissive license like BSD means this shouldn't be too much of an issue, though?

omasanori, 2024-08-27:
> I wonder what constitutes "useful"? :)

I meant platform-dependent, non-legacy part, probably. I didn't consider it tbh.