littlekernel/lk #410: compiling VisionFive2 failed

ILoveStartFive, 2024-06-15:
make[1]: Entering directory '/home/good/share/lk-master_new'
make[2]: Entering directory '/home/good/share/lk-master_new'
RISCV: Supervisor Mode
RISCV: MMU sv39
RISCV_EXTENSION_LIST = zba zbb
MISA_SPEC = 
make[2]: riscv64-elf-gcc: Command not found
LIBGCC = 
ARCH_COMPILEFLAGS = -march=rv64imafdc_zba_zbb -mabi=lp64d -mcmodel=medany
LINKER_TYPE=bfd
COMPILER_TYPE=gcc
Checking if -Wnonnull-compare is supported: no
PROJECT = visionfive2-test
PLATFORM = jh7110
TARGET = visionfive2
ARCH = riscv
TOOLCHAIN_PREFIX = riscv64-elf-
DEBUG = 2
including app app/shell app/stringtests app/tests arch arch/test dev dev/bus/pci dev/bus/pci/drivers dev/interrupt/riscv_plic kernel lib/aes lib/aes/test lib/cbuf lib/cksum lib/debugcommands lib/fdt lib/fdtwalk lib/libcpp lib/unittest lib/version platform target
including dev/net/e1000 kernel/vm lib/console lib/debug lib/heap lib/libc
including lib/heap/dlmalloc lib/io lib/libc/test lib/minip
including lib/iovec lib/pool
LIBGCC = 
GLOBAL_COMPILEFLAGS = -g -include ./build-visionfive2-test/config.h -Wextra -Wall -Werror=return-type -Wshadow -Wdouble-promotion -Wno-multichar -Wno-unused-parameter -Wno-unused-function -Wno-unused-label -fno-common -ffreestanding
GLOBAL_OPTFLAGS = -O2
ARCH_COMPILEFLAGS = -march=rv64imafdc_zba_zbb -mabi=lp64d -mcmodel=medany
ARCH_COMPILEFLAGS_FLOAT = 
ARCH_COMPILEFLAGS_NOFLOAT = 
generating build-visionfive2-test/platform/jh7110/module_config.h
generating build-visionfive2-test/config.h
build-visionfive2-test/config.h.tmp build-visionfive2-test/config.h differ: byte 3062, line 71
compiling platform/jh7110/platform.c
make[2]: riscv64-elf-gcc: Command not found
make/compile.mk:70: recipe for target 'build-visionfive2-test/platform/jh7110/platform.c.o' failed
make[2]: *** [build-visionfive2-test/platform/jh7110/platform.c.o] Error 127
make[2]: Leaving directory '/home/good/share/lk-master_new'
engine.mk:34: recipe for target 'make-make' failed
make[1]: *** [make-make] Error 2
make[1]: Leaving directory '/home/good/share/lk-master_new'
makefile:34: recipe for target '_top' failed
make: *** [_top] Error 2

travisg, 2024-06-17:
You'll need to have a riscv64-elf toolchain in your path. Depending on what
host OS you're using you can grab a precompiled one from
https://newos.org/toolchains/, such as
https://newos.org/toolchains/riscv64-elf-14.1.0-Linux-x86_64.tar.xz.

What host OS are you compiling on?

travisg, 2024-06-17:
Ah interesting, the comment in the email and what was pasted here is not the same. In this case it's compiling but the slightly older (gcc 12.1) compiler isn't recognizing the stimecmp instruction. Hmmm, I can have it use the hex version of the CSR, or I'd recommend getting a newer toolchain.

closed 2024-06-19 by travisg