littlekernel/lk #424: Searching for MSM8974 compiler.

sephiroth-mehdi, 2024-10-30:
For a special project, a special bootable compilation process needs to be provided for the MSM8974 chip. The information I have;

Little Kernel
Os + Radio Dump
qcm8974 Kernel
qcm8974 Device Tree

A master is wanted to compile the relevant product with a special service fee.
mehdieker123@hotmail.com

travisg, 2024-12-07:
Alas this really has nothing at all to do with this repository, since Qualcomm forked LK back in 2018 and has diverged into a completely different direction.

closed 2024-12-07 by travisg

sephiroth-mehdi, 2024-12-07:
Travis. I need you. I am having compilation issues for my MSM8974 device please help. 80% complete I am having include/path issues.
________________________________
Gönderen: Travis Geiselbrecht ***@***.***>
Gönderildi: 7 Aralık 2024 Cumartesi 10:54
Kime: littlekernel/lk ***@***.***>
Bilgi: MEHDİ EKER ***@***.***>; Author ***@***.***>
Konu: Re: [littlekernel/lk] Searching for MSM8974 compiler. / Special Service Fee. (Issue #424)

Alas this really has nothing at all to do with this repository, since Qualcomm forked LK back in 2018 and has diverged into a completely different direction.

sephiroth-mehdi, 2024-12-07:
If I send you the link, can you check the compilation? All files are ready but I am having minor problems like include <gsbi.h> during compilation.