littlekernel/lk #426: how to build with clang?

piterzhang, 2024-12-23:
The README mentions using gcc to build the project, but many projects nowadays use clang for compilation. Personally, I have always used gcc, and I believe many others might also be unfamiliar with building the project using clang. It would be helpful if the README could include detailed instructions or a guide on how to build the project with clang.

travisg, 2024-12-23:
Good idea! It's a bit mechanical right now, it needs to be a little smoother to be honest, which is when I was hoping to document it. But one way or another it should be written down.

travisg, 2025-01-07:
@arichardson If you have some time can you update some docs about how to set this up? I'd like to eventually have some sort of smarter USE_CLANG or whatnot switch, but for now it's a little finiky to set up clang, so would be helpful to add to probaly the top level README.md for now.

arichardson, 2025-01-08:
> @arichardson If you have some time can you update some docs about how to set this up? I'd like to eventually have some sort of smarter USE_CLANG or whatnot switch, but for now it's a little finiky to set up clang, so would be helpful to add to probaly the top level README.md for now.

I've submitted https://github.com/littlekernel/lk/pull/431 which documents how I am building. I agree that having a simpler variable such as `USE_LLVM=/opt/llvm/bin` would make this easier.

closed 2025-01-10 by travisg

travisg, 2025-01-10:
Excellent! Next thing I need to do is make it cleaner by adding some sort of switch to the build system to select it in a slightly smarter way than having to set all of the fields as in the docs.