littlekernel/lk #441: atol doesn't handle leading whitespace, and preceding '+'

HBehrens, 2025-04-08:
Thank for sharing your implementation of `atol` with a permissive MIT license! I ran this implementation against my test suite and found that a few edge-cases weren't handled the same way glibc does it. Further, when compiled against the target armv7, cases that produce an overflow led to illegal instructions.

I made the necessary minor adjustments to pass the tests (see below). Before I go through the hoops of creating a PR, please let me know if you are interested in merging this (at the time of writing, I see 24 open PRs).

```
static long atol_(const char *num) {
    long long value = 0;   // 👈  long long to handle overflow scenario
    int neg = 0;

    // skip leading whitespace   // 👈  leading whitespace
    while (isspace(num[0])) {
        num++;
    }

    if (num[0] == '0' && num[1] == 'x') {
        // hex
        num += 2;
        while (*num && isxdigit(*num))
            value = value * 16 + hexval(*num++);
    } else {
        // decimal
        if (num[0] == '-') {
            neg = 1;
            num++;
        }
        if (num[0] == '+') {    // 👈 preceding '+'
            num++;
        }

        while (*num && isdigit(*num))
            value = value * 10 + *num++  - '0';
    }

    if (neg)
        value = -value;

    return (long)value;
}
```

travisg, 2025-04-09:
Ah indeed. That does look good. A PR would be lovely!

re: the open PRs I really should close a bunch of those. Most are in some state where either the original author has wandered off or it can't get merged for one reason or another.

HBehrens, 2025-04-09:
Looking at this code again now, I think there's a bug with the newly introduced `num[0] == '+'` branch if the input starts with both a `'-'` followed by a `'+'`… Will fix this as part of a PR. Let me see how easy it is to set up the environment for this ;)

HBehrens, 2025-04-09:
For reference, this is the test I am running in my (unfortunately closed-source) project against `atoi` in which I am internally now calling the `atol()` function from above:

```
TEST test_atoi(void) {
    ASSERT_EQ_FMT(4, sizeof(int), "%d");
#ifdef PTR32
    // atoi and atol is really the same on Pebble's armv7 target
    ASSERT_EQ_FMT(4, sizeof(long), "%d");
    // but atoll returns an i64, so we can use that to force overflows
    ASSERT_EQ_FMT(8, sizeof(long long), "%d");
#endif

    #define AEQ(expected, actual) ASSERT_EQ_FMT((expected), (actual) , "%d")

    // initially generated via
    // https://chatgpt.com/c/67c8d6da-f6ac-800e-9495-1b68ec01578e

    // Basic cases
    AEQ(123, atoi("123"));
    AEQ(-123, atoi("-123"));
    AEQ(0, atoi("0"));
    AEQ(42, atoi("42"));
    AEQ(-42, atoi("-42"));

    // Whitespace handling
    AEQ(56, atoi("   56"));
    AEQ(78, atoi("\t\n 78"));
    AEQ(99, atoi("99   "));

    // Mixed input with non-numeric characters
    AEQ(123, atoi("123abc"));
    AEQ(-45, atoi("-45xyz"));
    AEQ(7, atoi("7.89"));
    AEQ(12, atoi("12e3"));

    // Empty and invalid input
    AEQ(0, atoi(""));
    AEQ(0, atoi("   "));
    AEQ(0, atoi("abc"));
    AEQ(0, atoi("+"));
    AEQ(0, atoi("-"));
    AEQ(0, atoi("--123"));
    AEQ(0, atoi("++123"));
    AEQ(0, atoi("-+123"));
    AEQ(0, atoi("+-123"));

    // Leading plus sign
    AEQ(123, atoi("+123"));
    AEQ(456, atoi("  +456"));

    // Edge cases
    AEQ(2147483647, atoi("2147483647"));
    AEQ(-2147483648, atoi("-2147483648"));

    // undefined behavior, Pebble-specific
    AEQ(-2147483648, atoi("2147483648"));
    AEQ(-2147483647, atoi("2147483649"));
    AEQ(2147483647, atoi("-2147483649"));
    AEQ(2147483646, atoi("-2147483650"));

    // both the Pebble runtime and Gossan's C-stdlib implementation crash here
    //    AEQ(0, atoi(NULL));

    PASS();
}
```

HBehrens, 2025-04-09:
I created https://github.com/littlekernel/lk/pull/442

closed 2025-04-09 by HBehrens